hd-zero/hdzero-vtx #1: Flashing blue led on vtx to indicate its in pit mode

oldrootbeer, 2022-07-12:
Any chance one of the Led's could flash at say 1hz if its in pit mode?

ligenxxxx, 2022-08-03:
@oldrootbeer Prior to this, if the VTX overheat protection was activated, the led flashed at 1Hz, the comments you made conflicted with this, so I tend to flash at 0.5Hz or 2Hz, which do you think is more appropriate?

oldrootbeer, 2022-08-03:
I guess 2Hz? Seems a little more noticeable in daylight.

ligenxxxx, 2022-08-04:
According to the priority, the status of the blue LED is as follows:
- After power up, when detected by msp_displayport, the LED flashes 8 times quickly and then lights up.
- When a camera is detected, the led is on, otherwise the led is off.
- When overheating protection, the LED flashes at 1Hz (the VTX_L LED will be off).
- When pitmode, the LED blinks at 2Hz.

ligenxxxx, 2022-08-25:
@oldrootbeer https://github.com/hd-zero/hdzero-vtx/actions/runs/2925528901 please test it.

oldrootbeer, 2022-08-25:
Thank you ligen. I will do when back home in a couple of days from vacation

oldrootbeer, 2022-08-28:
I've tested today. When I try to enter pit mode with stick commands (down and out), it enters pit mode for maybe 1/2 a second before coming straight out to normal power mode.
The blue led does not flash at all. Maybe because it is not entering pitmode with stick commands.

Hdzero freestyle vtx
Foxeer digisight v3
Rush Blade F7 HD FC
Betaflight 4.3.0

ligenxxxx, 2022-08-29:
I will check today.

ligenxxxx, 2022-08-29:
@oldrootbeer **https://github.com/hd-zero/hdzero-vtx/actions/runs/2947026030**
- Led flash as 2Hz has been fixed.
- I had the same problem with freestyle VTX Race v2 VTX no problem. Plan to debug.

closed 2022-09-01 by ligenxxxx