hd-zero/hdzero-vtx #2: Use PlatformIO

brandonrc, 2022-07-12:
I read that you are looking at replacing Keil IDE with an opensource alternative. 

I believe SDCC is supported with platformIO. 

Also it is easy for new developers to get hacking away and contributing to the project without a huge learning curve from a custom IDE like STM/Microchip/others are known for.

saidinesh5, 2022-07-13:
I'd say i am 30-35% there, trying to compile this with sdcc(4.2).
I have added a basic Makefile and I am able to compile the sources - I think.
I am currently trying to get the compiled sources to link and create a binary. I haven't done much baremetal programming in C, so I may need to read up on more things to get the linking working, Until then, feel free to review/help me out with the steps.
https://github.com/saidinesh5/hdzero-vtx/tree/sdcc

bkleiner, 2022-07-13:
i created a basic platfomio project here:
https://github.com/bkleiner/hdzero-vtx/tree/platformio

the pio part it does what it should - however there are still a bunch of issues with using sdcc to compile a flash-able binary

SumolX, 2022-07-13:
I have a compiling SDCC version of this software and I'm also using cmake.  All that's missing is the conversion from A51 pseudo code to bootloader asm.

https://github.com/SumolX/hdzero-vtx/tree/cmake_support

bkleiner, 2022-07-13:
retaining compatibility with keil is an interesting angle, however i suspect to take advantage of uVisions debug capabilities they would want to maintain their project files _either way_ so i don't know if cmake actually provides any benefit there.

you might wanna join us on the hdzero discord https://discord.com/invite/kGsnEDMb2V?fbclid=IwAR2eb9m5YUIOLAl5Nq5bY_tuq94awmhvpPFlBz0jKxbktXHVmIOgkSqY-Gw ("open-source" channel)

there are a couple of people there figuring stuff out together.

SumolX, 2022-07-13:
I use cmake because that is what I'm accustomed too.  I am able to build this source both Linux and macOS.... having windows support would probably be an easy port as well.

KarateBrot, 2022-07-13:
@SumolX Once building works on Linux you can always use the WSL to also compile on Windows. But having full win support would be the cherry on top.

bkleiner, 2022-07-13:
PIO comes with seamless cross platform support out of the box ;)

SumolX, 2022-07-13:
I will have to check it out.  As long as the code compiles and produces a valid binary.... cmake can always be replaced.

closed 2022-07-25 by ligenxxxx