hd-zero/hdzero-vtx #14: mspVTX issue

ligenxxxx, 2022-07-22:
### Ryan(discord) report:

1. Sometimes when I arm the drone, the video goes to rainbow then back to my desired power level - it is a split second

2. (probably a betaflight issue) if smart audio is enabled in the ports tab, then msp-vtx doesn't work

3. This video shows that VTX turns on immediately on boot rather than waiting for FC to assign channel and power level (I need the VTX to boot up at 0mW!)
https://1drv.ms/v/s!AibfK-JCIxFUjfYRoMr5nj46T1OTow?e=Ste5f1

4. When I arm the drone after enabling video via radio switch, the video flashes off then on for a split second
https://1drv.ms/v/s!AibfK-JCIxFUjfYSJZ5nf7boY_kMbQ?e=pxaeVv

5. Video turns off after takeoff - very dangerous!
https://1drv.ms/v/s!AibfK-JCIxFUjfYQJNkI5LhFXp6uGw?e=IpoiUq

To reproduce the issue, make sure that VTX power is controlled by a switch on the radio. I set my switch up so that power is 0mW by default and I must flip the switch from default position for the vtx power to come on

vtx 0 3 0 0 3 900 1300
vtx 1 3 0 0 1 1300 1700
vtx 2 3 0 0 1 1700 2100

### aviphysics(discord) report:
"the power low until first arm" option doesn't stick. That seems like a BF issue.

rquellet, 2022-07-22:
Looking forward to these issues being resolved

bkleiner, 2022-07-22:
1, 3, 4 and seem related to switching the rf-mode, causing it to lose vdet. 
5 is one of those "could be everything" type issue.

ligenxxxx, 2022-07-25:
I didn't consider the third point at the time. I wanted to solve this problem first. VTX needs to initialize RF after detecting mspVTX.

closed 2022-08-03 by ligenxxxx