hd-zero/hdzero-vtx #21: VTX Menu Navigation is Slow

rquellet, 2022-07-31:
Navigating the VTX settings menu is very slow. It seems I must hold my transmitter stick for 0.5sec before an action is registered

llambkin, 2022-08-03:
Hey mate. What VTX is that? I have a new whoop lite board that refuses to go into the VTX menus altogether. Canvas mode is working and OSD elements are showing,I can edit the PIDS and rates etc, but the VTX stick commands are not reaching the VTX at all. I did not connect the SA pin to the FC, and the TX and RX pads are connected correctly. Can't access the cam menus either.

constanan, 2022-08-04:
Same for me. My board is BETAFPV F4 1S AIO 5A with ELRS 2.4 GHZ refuses to go into the VTX menus . Canvas mode is working and OSD elements are showing,I can edit the PIDS and rates etc, but the VTX stick commands are not reaching the VTX at all. I did not connect the SA pin to the FC, and the TX and RX pads are connected correctly. Can't access the cam menus either. Any fix for that?

ligenxxxx, 2022-08-09:
@llambkin @constanan According to your description, I guess FC may not receive the data sent by VTX. Can you check the second page of the VRX menu for any error messages?

ligenxxxx, 2022-08-09:
@rquellet I suspect this is due to the large delay in the OSD display. More related to VRX, this is an item to be optimized.

ligenxxxx, 2022-08-15:
@rquellet Using bytefrost vrx to view the OSD, there is no problem with refresh rate and response speed.
I can't think of anything that could be optimized on the VTX side for now. So I'm closing this issue.
I plan to fix this in HDZero goggle first.

closed 2022-08-18 by ligenxxxx

llambkin, 2022-08-20:
No need to reopen this issue, just updating my status on my particular issue. I gained access to the VTX menu by swapping the VTX UART with the receiver UART. In my case, the VTX menus were inaccesible on UART1, when I moved it from UART 1 to UART2, both the VTX menus and the receiver (with telemetry) work correctly.