hd-zero/hdzero-vtx #23: Remove Redundant Camera Settings Function

MacDaddyFPV, 2022-08-03:
Currently for each camera type there is an auto and manual setting. Whilst in the camera menu, the first option is to scroll through the different camera type settings. For Auto, the setting is named as per the camera type and the settings are default. None of the settings besides HV flip can be changed when this is selected. For Manual, the setting is named as per the camera type with an "M" at the end and the settings all start at the lowest (eg brightness is 0 and saturation is lvl 1). All of the available settings can be changed when this is selected.

I believe having these different settings will become confusing for the users and I can not see it providing any benefit.

I recommend having a single settings profile per camera type and have the default set to the best know settings for that camera.

ligenxxxx, 2022-08-12:
https://github.com/hd-zero/hdzero-vtx/actions/runs/2846284976
Please check it.

MacDaddyFPV, 2022-08-13:
@ligenxxxx I have tested with the Micro V1 on the Whoop board and the Nano Lite on the Whoop Lite.
Both functioned well and all settings were able to be changed as expected. Having just the single option for each camera feels much cleaner. Thanks!

closed 2022-08-14 by ligenxxxx