hd-zero/hdzero-vtx #24: Broadcast "No MIPI" text at 0.1mW when MIPI is not detected

sunolfpv, 2022-08-04:
I ran into a small problem for the first time today. I had a bad/partially attached MIPI cable, so the VTX wouldn't broadcast. However, the MIPI cable was apparently good enough that the blue LED light came on, which made me think the MIPI was fine and the VTX was bad. Eventually switched out the cable and all good now.

Suggest when the VTX doesn't broadcast because of a MIPI situation, it broadcasts out "No MIPI" at pit mode level (0.1mW) so the user knows what the issue is.
Sending out error codes at low wattage may be helpful in other circumstances.
[image]

MadRC, 2022-08-07:
I really like this idea, it would help to be able to diagnose issues with VTX for users. Perhaps we could take this a step further and also overlay other diagnostic data live firmware version and may be some boot checks.

ligenxxxx, 2022-08-10:
https://github.com/hd-zero/hdzero-vtx/pull/25
Can you try this?
When no mipi signal is detected, the blue light will turn off.

sunolfpv, 2022-08-10:
Thanks and I can confirm that the board behaves like that (blue light off when no MIPI) MOST of the time. However, since I opened this I found I can create a situation where everything looks normal and the blue light was on, but the board was not broadcasting anything (or at least nothing the RX can lock on).  A partially or loosely connected MIPI will do this. Blue light, no video.

So, there is fixing that, but I think the bigger opportunity would be text details of any error condition if possible.

The attached picture shows blue light and a small signal, but nothing the RX can lock on to. Other times the RX won't show any signal, but the blue light is on. Both situations could lead to users thinking the board was bad.

[image]

sunolfpv, 2022-08-10:
I just noticed a review on NewBeeDrone where someone left 1 out of 5 stars on the Whoop Lite Bundle because they thought the VTX was DOA. It really looks like user error, but it's another example of where a "No MIPI" low wattage broadcast could be handy.
I left a review on NewBeeDrone to try to point out the user error.
https://newbeedrone.com/collections/hd-shark-byte-gear/products/hdzero-whoop-lite-bundle

ligenxxxx, 2022-08-11:
I changed the default state of the blue LED to be on after power on, no matter what the camera is, at least I can see if the 5680 is working.

Yith1, 2022-08-23:
I had this exact problem when one single wire on the mipi connector was cut during a crash. The blue light was on solid, but the VTX was not transmitting at all. It had me scratching my head.

The one spare part I had to hand was a mipi cable so I swapped that in and it all started to work again. 

So yes, some sort of visual aid for this problem would be helpful.

oldrootbeer, 2022-09-02:
Agree @Yith1. I've had this yesterday with a rouge bit of uhu por glue on the camera mipi connector. Solid blue, Spectro shows grey at the specified channel at low power but no green bar and no image, and a spread of lower power across the board.

closed 2023-05-24 by ligenxxxx