hd-zero/hdzero-vtx #36: Redefine the version number rule

ligenxxxx, 2022-08-18:


bkleiner, 2022-08-19:
could you iterate a bit on what you would see here?

what i personally found to work well with my opensource projects:
firstly _use semver_, be liberal with new version numbers.
semvers can pretty nicely be packed into a uint32_t making them well suited to be used in source code too.

- master contains the latest _stable_ version
- a separate branch called "develop" is used to merge all ongoing work into
- whenever a stable state is reached develop is merged into master and a new release is tagged on github
- `-dev`/Release candidate versions are tagged on the develop branch, leading up to stable releases on the master branch

Yith1, 2022-08-24:
Whatever is decided then it needs to be consistently used in the VRX display, the Downloads page on the website, and the file names, including ideally the actual firmware binary files themselves.

Right now it is rather difficult to compare these as the files and web page use dates, the VRX display uses an odd sequence of version digits and the firmware binaries have no version numbers on them at all.

Also it would be good if the VRX could display the firmware version of the VTX it is currently receiving data from, though I expect that to be a lot more difficult to do.

ligenxxxx, 2022-08-25:
@Yith1 
- Currently, VRX only recognizes firmware with a specific filename and updates it. At the beginning of the design, this was simple to implement. It should be possible to add a version number to the firmware name, as long as certain rules are met.
- Displaying VTX's version number in VRX is not difficult, it's worth trying but not now.
Most of the above two points require VRX to do it, I think I'll take your input.

closed 2022-09-04 by ligenxxxx