hd-zero/hdzero-vtx #38: Advanced OSD

Remleyfpv, 2022-08-19:
I thought this might be a very cool feature. A color OSD with special features like: Battery turning red once it gets low, A red warning indicator, maybe even when there is a warning for example not enough sat's It could just say warning in the midddle of the screen and then flash the sat indicator in red (where you have it setup in your osd)

ligenxxxx, 2022-08-27:
I plan to implement a custom OSD in goggle first.

Remleyfpv, 2022-08-27:
> I plan to implement a custom OSD in goggle first.
Do you mean an osd for like goggle battery and stuff?

ligenxxxx, 2022-08-30:
I mean FC OSD.

RBiggerstaff, 2023-03-13:
I am using the Matak F411 WTE with iNav 5.1, HD Zero Rx, HD Zero Freestyle Tx, and Runcam HD Zero Micro HD camera.  I am powering the TX on a separate bat w/capacitor but the rest is wired to the FC as shown in the manual for the Tx and FC (RX to TX and TX to RX - not using SA).  I have update HD Zero firmware twice.   I have watched as many vids as I could find on setting this up for OSD so I have "HD" chosen for video and "HD Zero" for UART1 (SRXL on UART2) with the TX plugged into UART2 on FC (I learned not to use SoftSeriel for this).  I am getting HD video but no OSD.  Any suggestions?

closed 2023-05-24 by ligenxxxx