hd-zero/hdzero-vtx #52: Feature request to update vtx firmware through usb interface

kakashi2018, 2022-10-23:
Hi, is it possible to update the vtx modules directly through usb from a computer. With new devices like frsky x20hd which supports hdzero lacks upgrade port for vtx and if the radio is the only thing that one is having like me there is no way to upgdate the firmware on vtx as of now.

KarateBrot, 2022-10-23:
That's possible via Betaflight's USB passthrough.

But afaik FrSky has to implement it into their flashing app, if they want to support it. Same for HDZero.

kakashi2018, 2022-10-24:
> That's possible via Betaflight's USB passthrough feature.
> 
> But afaik FrSky has to implement it into their flashing app, if they want to support it. Same for HDZero.

Can you please explain how hdzero vtx firmware update is done via betaflight passthrough or any link related to it?

KarateBrot, 2022-10-24:
I have no idea. You should ask HDZero about that. They need to make it happen.

ligenxxxx, 2022-10-24:
https://github.com/bkleiner/hdzero-flasher This might be useful if you have programming background.
Frsky should fix that. All VRXs should have the ability to update the VTX.

bkleiner, 2022-12-20:
It appears frsky has started shipping usb flashers. Closing.

closed 2022-12-20 by bkleiner