hd-zero/hdzero-vtx #58: [Request] Could you make a simple libary for Arduino to send OSD to HDZero?

human890209, 2022-11-08:
Hi. It's nice to have the Betaflight OSD in HDZero. I wish it could be easier to display the OSD with Arduino. 
I'm trying to replace my analog VTX/VRX on my arduino robot with HDZero but I fail to understand how the MSP OSD work in Betaflight.  I used to use AT7456E to display voltage, status, data and menu on the analog screen.
An MSP OSD library with example, or a simple arduino example project is nice to have. Wish HDZero could be popular in the Arduino Community.

ligenxxxx, 2022-11-09:
If you are interested in implementing it yourself, please join our discord channel where some people know about this. We are willing to help with this.
https://discord.gg/SQnq73G5Gs

ligenxxxx, 2022-11-23:
@human890209 https://github.com/hd-zero/hdzero-vtx-arduino

closed 2022-12-10 by ligenxxxx