hd-zero/hdzero-vtx #59: Unlocked Freestyle VTX on MSP VTX doesn´t put MAX frequency on VTX Table

zandorsp, 2022-11-12:
I'm using Unlocked Freestyle VTX on latest master ( november.11.2022) on latested Betaflight master too (4.4) using the MSP VTX funcion.

The VTX Table is incorrectly populated: 

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY    0    0    0    0    0    0    0    0
vtxtable band 2 BOSCAM_B B FACTORY    0    0    0    0    0    0    0    0
vtxtable band 3 BOSCAM_E E FACTORY    0    0    0    0    0    0    0    0
vtxtable band 4 FATSHARK F FACTORY    0 5760    0 5800    0    0    0    0
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I FACTORY    0    0    0    0    0    0    0    0
vtxtable powerlevels 5
vtxtable powervalues 14 23 0 0 0
vtxtable powerlabels 25 200 0 0 0

The expected is:
# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 0 0 0 0 0 0 0 0
vtxtable band 2 BOSCAM_B B FACTORY 0 0 0 0 0 0 0 0
vtxtable band 3 BOSCAM_E E FACTORY 0 0 0 0 0 0 0 0
vtxtable band 4 FATSHARK F FACTORY 0 5760 0 5800 0 0 0 0
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6 I CUSTOM 0 0 0 0 0 0 0 0
vtxtable powerlevels 5
vtxtable powervalues 14 23 27 30 0
vtxtable powerlabels 25 200 500 MAX 0

zandorsp, 2022-11-19:
The wrong value sent to flight controller's vtx table is:
vtxtable powervalues 14 23 0 0 0
vtxtable powerlabels 25 200 0 0 0

The correct and expected is:
vtxtable powervalues 14 23 27 30 0
vtxtable powerlabels 25 200 500 MAX 0

ligenxxxx, 2022-11-21:
Is the vtx table set correctly when SmartAudio is not enabled?

ligenxxxx, 2022-11-21:
[hdzero_freestyle.zip](https://github.com/hd-zero/hdzero-vtx/files/10054931/hdzero_freestyle.zip)

zandorsp, 2022-11-23:
> @zandorsp [hdzero_freestyle.zip](https://github.com/hd-zero/hdzero-vtx/files/10054931/hdzero_freestyle.zip)

I tested with sucess :)
vtxtable powerlevels 5
vtxtable powervalues 14 23 27 30 0
vtxtable powerlabels 25 200 500 MAX 0

closed 2022-11-25 by ligenxxxx