hd-zero/hdzero-vtx #73: Camera image cuts every few seconds on FreestyleVTX.Rev22122022

oldrootbeer, 2022-12-23:
I have updated my sb vrx to 22122022. New fonts are visible.
When my freestyle vtx's are updated to 22122022, the video cuts every few seconds. Almost as if its losing the rf centre frequency periodically. Video - https://www.youtube.com/watch?v=xphZvGRcftI

The video is ok when the vtx is at 11062022.

oldrootbeer, 2022-12-30:
hdzero-vtx build #123 works for freestyle & digisight v3

ligenxxxx, 2022-12-31:
#123 is the latest version for this issue.
I see you mentioned no video in PR#76, actually outdated information, right?

oldrootbeer, 2022-12-31:
> #123 is the latest version for this issue. I see you mentioned no video in PR#76, actually outdated information, right?

This is right. Sorry, still finding my way around github.