hd-zero/hdzero-vtx #78: OSD Refresh rate.

lesharodin, 2022-12-29:
Hi. What is maximum refresh rate HdZero is capable of?
Now its only updates 3 times per second (when I add milisecond timer to OSD).
Maybe we need to define MSP osd elemets and preoritize some for fast updates (like horizon and timers), and some like osd names and channels could update less frequently?

rquellet, 2022-12-29:
It's capable of 10hz typically. Recent FW updates have slowed things a bit on the VRX device. HDZero goggles has a faster processor and refresh rate is higher. Ligen has some experimental code that gets OSD refresh up to 20-30hz on the HDZero goggles

lesharodin, 2022-12-30:
Hmm I dont think we have 10hz now, more likely 3hz.
Sad we dont have a VRX code yet to take a look.

lesharodin, 2023-02-19:
> Currently only hdzero goggle can flash OSD quickly. As for how to increase the refresh rate of other sb vrx/vrx4 etc., it is still in the experiment and needs a lot of time.

Will VRX FW be opensource at some time for community to try to fix this issues?

closed 2023-05-24 by ligenxxxx