hd-zero/hdzero-vtx #83: High contrast video every ~5 power cycles or so on whoop lite vtx..

Tbell123, 2023-01-08:
On the whoop lite vtx seeing an extremely high contrast video feed every ~5 power cycles. Its looks like the contrast settings are being messed up, but then power cycling makes it go back to normal. This is using the whoop camera on a 65mm 1s whoop.

[image]
[image]

FractalEngineer, 2023-01-08:
Contributing a dvr of my similar issue https://youtu.be/TnTCVvRoCfo

gizlonk, 2023-01-08:
I too am experiencing this same issue. 4-5 packs and the image becomes extremely high contrast. Next cycle and it's OK.

iz3man, 2023-01-16:
Got that issue too. Was wondering if some settings of the camera were altered. Reconnected the battery and everything was fine again. I was assuming an almost empty battery being the cause. But this doesn't seem to be the case.

jgauruder, 2023-01-23:
I have the same symptom...indoor 65mm whoop lite vtx build with nano lite cam.  thought maybe I was just powering up while pointing the camera directly at a light in my kitchen or something was causing it randomly to freak out the camera sensor.

ligenxxxx, 2023-01-23:
DId you try v1.1

获取Outlook for Android<https://aka.ms/AAb9ysg>

jgauruder, 2023-01-23:
@ligenxxxx I will hopefully be able to try that tonight.  Thank you!

jgauruder, 2023-01-24:
After flashing, I only observed the high contrast once so far.  So not completely eliminated, but so far for me  it seems less frequent than before.  I will see if I can check my DVR to see how many power up/on of vtx was made since update.  I'm guessing maybe 10 total as I tinkered with osd object placement tuning after the firmware update.

gizlonk, 2023-01-24:
> After flashing, I only observed the high contrast once so far. So not completely eliminated, but so far for me it seems less frequent than before. I will see if I can check my DVR to see how many [...]

This happened to me too. 2nd pack had "drunken camera" and none since then. Worth noting that the camera MAY have been pointed to the light in the room when plugging the battery in.

ligenxxxx, 2023-01-28:
https://github.com/hd-zero/hdzero-vtx/releases/tag/v1.1.0

closed 2023-03-12 by ligenxxxx