hd-zero/hdzero-vtx #85: Feature request: Irc tramp protocol

timfredriksson, 2023-01-09:
It would be wonderful if the vtx could support the irc tramp protocol for us ghost users. It is a feature that is working so awesome in the ghost ecosystem and I miss that I could change channel on the back of the radio a lot.

pidschey, 2023-01-10:
And one step further a opportunity for ghost rx backpack in the google, that the vtx channel will changed automatically in the google, when changing channel changing by ghost tx

ligenxxxx, 2023-05-24:
- We are adding the irc tramp protocol.
- Elrs backpack and ghost system do not seem to be universal.So goggle cannot switch channels automatically.

ligenxxxx, 2023-09-21:
The IRC protocol has been implemented, but it requires VTX hardware support and special ghost fw.

closed 2023-09-21 by ligenxxxx