hd-zero/hdzero-vtx #90: Feature Request: Shortcut to 25mw

sabshire, 2023-01-16:
It would be cool to have a shortcut to enter/exit 25mw mode.

The LP_Mode set to 1st is great, but I like to talk to the camera after flying (and possibly multiple disarms), so it would be nice to be able to quickly set the power level to 25mw to help reduce the interference when the drone is close to the goggles.   Since I fly in areas where I don't always want 25mw mode after disarm, it would be nice to have a shortcut to get there without having to go to the vtx menu, and change the power setting, and then change it back.

bkleiner, 2023-01-16:
Aren't there multiple ways to achieve this?
- [power level on a switch](https://github.com/betaflight/betaflight/wiki/VTX-CLI-Settings#change-vtx-power-level-using-aux-channel) via SA
- [power level on a switch](https://github.com/betaflight/betaflight/wiki/VTX-CLI-Settings#change-vtx-power-level-using-aux-channel) via MSP-VTX
- power level via [elrs backpack](https://www.expresslrs.org/2.0/hardware/backpack/esp-backpack/) (lua script)

really feels like this something that should be externally driven.

sabshire, 2023-01-16:
I don't use ELRS, or SA.  Not sure how you would do option #2...  care to elaborate, as a switch to change it would be a nice option.

bkleiner, 2023-01-16:
humm, we are indeed lacking good resources on the topic of MSP VTX.
there is [this video](https://www.youtube.com/watch?v=-el0OhdBhU4) but it covers a bunch of other stuff and is slightly outdated already.

The tldr is you need BF 4.4 and load the preset 
[image]

sabshire, 2023-01-16:
Hmmmm seems like a lot of work and then I'll have to deal with re-tuning...
Yeah I'll skip updating to 4.4 to get that feature.

closed 2023-01-16 by bkleiner