hd-zero/hdzero-vtx #97: Feature Request: Better (more sensible) values for shutter

sabshire, 2023-01-25:
I asked on the discord, and got an answer about the shutter values that doesn't seem to make sense, so I went digging through the source, and was trying to decipher how the values of 1-32 are sent to the camera.  And though I figured out the values being sent and the addresses where those values are set in the camera, without camera documentation, they don't make much sense.  For example, why is auto sending 1120, but the fixed value range is from 1 (25) to 32 (800)?

From my discord conversation, they said 1 -> 0.37ms, and 32 -> 12.12ms..  I have no idea what that means.  Tinkering with the values in the camera, and testing didn't result in much of a difference, although at lower values the color of the image seemed to change a bit (less saturated) in certain lighting conditions.

It would be great to have values here that made more sense as to the actual shutter speed, and it would be even better if we had a broader range for setting a fixed shutter speed.  

At the very least, some insight into what the values mean, and if there is any difference for the user when setting them.  And if there is no practical difference for the user, it doesn't make sense to even have a setting.

mard89, 2023-01-27:
fps can easily be calculated from ms with the formula: fps = 1/(ms/1000). For instance:

**fps  -      ms**
60    -    16.667
90    -    11.11
120   -    8.333
2702   -  0.37
82    -     12.12

But "1/" before the fps numbers and you've got a shutter speed as used for still images. So 82 fps becomes 1/80

sabshire, 2023-01-27:
The shutter setting is not fps.

closed 2023-05-24 by ligenxxxx