hd-zero/hdzero-vtx #100: With LP_MODE on, VTX menu shows 25mw

evilC, 2023-02-01:
1. ELRS LUA script set to Pwr lvl 2 and LP mode off. VTX currently at 200mw power
2. Enter VTX menu (Sticks down and together)
4. Enable LP Mode
5. At this point, the menu shows Power: 200, LP Mode: ON
6. Exit VTX menu
7. Enter VTX menu
8. ERROR: Menu shows Power: 25, LP Mode ON

ie it appears to the user that when they arm, power will not be set to 200
When you do arm, it does go to 200mw

Fix: In the VTX menu, it should not show the CURRENT VTX power, it should show the power level that it WILL go to when you arm.

pitts-mo, 2023-08-02:
Hello evilC,
What you are describing sounds like the normal Betaflight/Inav low power disarm feature. Are you sure that has been turned off in your flight controller software and have you verified the flight controller is really commanding 200mw at the time the VTX menu is reporting 25mw?

evilC, 2023-08-02:
I am not saying that it really has commanded 200mw when it should be 25, I am saying that the UI kind of appears to say that it is.
ie it's more of a UX issue than a functional issue.
If you set it up to be 200mw on arm, and LP mode on, when you save and go back into the menu, it APPEARS as if you have set LP mode on and to only go to 25mw on arm, because power level is showing 25.
With LP mode on, it appears to be impossible what the CURRENT setting is, because the UI is incapable of showing that.
In an ideal world, rather than just having one value which shows current power is, there should be two settings: One to show current power level, and one to show what it will go to on arm. Either that, or it should show something like `25 (200)` to reflect that it is currently on 25, but will go to 200 on arm.
I checked and BF low power disarm was indeed on. I switched it off and the issue persists.

evilC, 2023-08-02:
Not wholly.
When you go into the menu and enable LP mode, then set the power level to 200, the menu shows:
POWER: 200
LP MODE: ON

This gives the user all the information that they need - power will be at 25mw when disarmed, and be raised to 200mw on arm.

If you save and exit the menu, then re-enter the menu, it now reads:
POWER: 25
LP MODE: ON

Nothing has changed. it still appears to go to 200mw on arm, but the menu is now giving spurious information. I suspect that what is going on is that when you ENTER the menu, the `POWER` line is showing what it is **currently** transmitting at, not what it will change to upon arm.

evilC, 2023-08-02:
[CORRECTION]
If you save and exit the menu, then re-enter the menu, it reads:
POWER: 200
LP MODE: ON

If you save and exit the menu, arm and dis-arm, then re-enter the menu, it reads:
POWER: 25
LP MODE: ON

pitts-mo, 2023-08-02:
Yes, This is occurring because the VTX is only aware of the current value and the potential list of values. And I do not see a way to implement a feature able to show the current power level VS desired power level. At least not without removing all the external control functions (eg: remove Smart Audio, Tramp, and MSP-VTX).

Sure, You can set a value and it will change but then it is being changed again and we can not guarantee what the resulting value of a change will be. The best place to turn for a feature that shows the desired value vs current value would be in the flight control software since it allows for an OSD that is available during "normal" use and it has it's own desired value. However, here too you run into the same problem... You can add more complexity and OSD clutter to attempt to track a desired value from a particular device but that still does not mean it will be the resulting power value at any point in time.

evilC, 2023-08-02:
> Yes, This is occurring because the VTX is only aware of the current value and the potential list of values. 
So how come if I exit the menu, then re-enter it without arming, it shows 200, even though the current value is 25?

pitts-mo, 2023-08-02:
The current output power value will remain current until a new output power is set.

ligenxxxx, 2023-09-21:
@evilC https://github.com/hd-zero/hdzero-vtx/pull/151
Please check this PR.

evilC, 2023-10-27:
@ligenxxxx apologies for the delay, I missed this. I just flashed it and will hopefully get to test this weekend, weather permitting

evilC, 2023-10-29:
@ligenxxxx 
I tested on Whoop Lite and my mate tested on his unlocked Freestyle v2. Both on LowBand
We only got 4 or 5 packs in each today, but it seems fine