hd-zero/hdzero-vtx #102: V1 Cameras "flashing" image

MacDaddyFPV, 2023-02-14:
On the nano/micro v1 there is a constant flashing of the image as if it is trying to change between day and night mode. This is mostly visible when going from sunny to darker areas.

T-M-0, 2023-02-14:
I can confirm this problem on a nano V1 cam.
It happens when I fly from the sun into the shade on a sunny day.
It starts with FW 06012023 or 22122022

mwasserm, 2023-02-15:
I also have this issue on a nano v1 camera with original whoop vtx. I noticed the issue after upgrading the vtx and goggle to version 06022023 along with a betaflight flash to 4.4 for the HD OSD. I don't recall the prior vtx firmware version, but it did not exhibit this behavior with the same camera/vtx/goggle combination along with betaflight 4.2. Here is a short video illustrating the issue: https://youtu.be/7xpLYOgoq9g. Several folks have made suggestions to mitigate this flashing, but it really does appear like this is a bug as it occurred only after updating the firmware. Happy to provide more detail or test things out if it would be helpful.

ligenxxxx, 2023-02-15:
I'm working on it. It takes a little time to debug.

ligenxxxx, 2023-02-15:
https://github.com/hd-zero/hdzero-vtx/actions/runs/4181809548

CyberFpv, 2023-02-15:
Got same problem with 3 nano v1 after update to 060123 for my new hdzero goggle.. white balance on or off same things.. flashing

ligenxxxx, 2023-02-16:
https://github.com/hd-zero/hdzero-vtx/actions/runs/4181959009
Please test the firmware in the link.

mwasserm, 2023-02-18:
Hi, thank you for such a quick turnaround on this! I only had a few minutes, but I flew before (still had flashes), then flew another battery after updating to this firmware. The flashing appears to be fixed. I'll try and fly again tomorrow to confirm, but so far it looks like it's fixed. I might be mis-remembering, but I still think the nano v1 still looks grainier in the shadows than I remember when I first got it last summer, but it's at least back to the performance immediately before the flashing started.

CyberFpv, 2023-02-18:
Yayyy.. installed now... and flashing gone.. problem fixed for my first
quad.. leter i test the other 2 cam v1... thank you

jsaginario, 2023-02-19:
Good for me as well, much improved. Thanks Ligen!

closed 2023-02-20 by ligenxxxx

csmicfool, 2023-03-11:
Confirm this fixed the issue for me too.

Thanks @ligenxxxx !!