hd-zero/hdzero-vtx #103: VTX off with stick command did not work on on FW 06022023

T-M-0, 2023-02-14:
With FW 06022023 I can´t switch the VTX off with stick command.
On FW 06012023 its working.

Tested on Freestyle VTX and Race V2

Smartaudio is not connected.
FC firmware is FETTEC Alpha.
I also tested a few versions. It does not influence the issue.

FIynix, 2023-03-16:
I can confirm the same problem is still present on firmware V1.1.0.

Using : 
VTX race V2
Camera nano 90
Betaflight 4.2.11 (hobbywing F7)
ELRS 3.1.2

ligenxxxx, 2023-03-17:
https://github.com/hd-zero/hdzero-vtx/actions/runs/4259879649

FIynix, 2023-03-17:
We need a way to activate/deactivate 0mw with stick command.
The 0mw stick command was allowing that. 0mw boot option won't work to close the vtx at the end of a race.

T-M-0, 2023-04-18:
Still not working with the latest firmware.
Tested it also with INAV 6.0.
The VTX does nothing with stickcommand.
Even in the manual it is described that it should work.

jissoe, 2023-06-20:
I can confirm this is still present on Rev 13042023 with raceV2 vtx. Whoop vtx is working fine.
I noticed the stickcommands are working again when connecting smartaudio.

pitts-mo, 2023-08-02:
Hello, 
Is this still an issue with VTX firmware Rev_26062023 / 1.3.0  ?

T-M-0, 2023-08-08:
Yes, its still an issue.
I think that will not be fixed.....

On a whoop board (not the light) it worked with the actual firmware.

Also I talked with the FETTEC programmers, and they say the implemented the MSP protocol like the specs.
MSP V2 actually not implemented.

There must be a little bug on the VTX firmware what causes this issue

ligenxxxx, 2023-09-21:
@T-M-0 Can this bug still be reproduced?

closed 2024-06-11 by ligenxxxx