hd-zero/hdzero-vtx #109: signal lost grandually then comes back

TwistedFPV, 2023-03-14:
signal will gradually get worse and be lost completely ,after a few seconds comes back and cycle repeats.
rebooting goggles fixes the issue.

https://www.youtube.com/watch?v=0pNWOFrVawU

after rebooting goggles
https://www.youtube.com/watch?v=dJQVkXnu1NQ

MeatG, 2023-05-24:
Yesterday, after lot of packs flown I got the same problem once with one of my race quads (race vtx 2) and hdzero goggles: gradually lost signal until it was almost completely lost 10-20m away while flying. I landed for a moment and suddenly the grain disappeared and all was good.

Feels like a false thermal throttling or something? It seems so "mathematically" gradual. Air temp was about 18'C and was flying fast so cooling shouldn't have been a problem. I have no dvr cause my goggles didn't record for some reason 🤪

evilC, 2023-05-25:
Yeah, I have seen something like this recently too

pitts-mo, 2023-08-02:
Hello,
Does this issue persist even if both VTX and VRX/Goggle are running current firmware Rev_26062023? 

Are there any other observations that appear to suggest a cause?