hd-zero/hdzero-vtx #116: iNav - HD Zero - Stick Command Issues

MartinHugh, 2023-04-04:
A common stick command in iNav is to save settings after tuning things like auto level, auto tune, auto trim.
This is normally done after landing and disarming.
It is useful to see confirmation that this has happened on the iNav OSD after saving.
If the video is being recorded (which it usually is) this can be checked later if necessary too.

However.  The same stick command used in INav to Save Settings, is now used by HDZero to set 0mW power.
This command works even if LP MODE and PIT MODE are OFF.

The result is that Save Settings in iNav simply records the video going to HDZero Rainbow display, rather than showing confirmation of iNav save.  It probably does Save Settings too, but who knows.  

A secondary issue (though not the point of the one being raised) is that trying to bring the vtx out of 0mW mode (sticks in and down) does not work. One has to cycle the power on the aircraft to get the video to come back).

This is all a bit of a mess.

Can we please allow 0mW to be put on a different stick gesture so that it does not conflict with iNav, or allow it to be disabled.

ligenxxxx, 2023-04-19:
`allow it to be disabled.` is a good idea.

ligenxxxx, 2023-09-21:
At present, the stick command of FC and VTX cannot be truly avoided. I'm still thinking of other options.

closed 2023-10-12 by ligenxxxx