hd-zero/hdzero-vtx #118: Improve video transmission diagnositics

MartinHugh, 2023-04-09:
Currently (as far as I can tell) the blue LED works as follows:

Eight fast flashes followed by solid on 
    = Normal operation

Flashing once a second 
  = Problem with MIPI or camera 
OR 
  = throttling power because of overheat.

No Blue LED 
  = Problem with MIPI or camera 
OR
  = incorrect firmware installed

It would be an improvement if we can make it easier to diagnose and distinguish between the above states which are very confused.  Especially where there a single state represents multiple possible problems, and where multiple states can represent the same problem.

I am not sure if Pit mode or Lower power mode flash to indicate when they are active, but I think such indications would be useful, but they would need to be distinguishable from everything else

ligenxxxx, 2023-04-22:
- 8次快速闪烁已经过时，最新的fw中只会闪烁3次。
- 我们需要一个wiki，当每次修改或者增加功能的同时更新wifi。
- 长脉冲和短脉冲混合，我喜欢这种方式。

AlexandreBonneau, 2023-06-02:
> 8 quick blinks is outdated, only 3 blinks in latest fw.

Then the PDF manual must be updated as well, because it currently provides false information.

About the wiki, it could be created either in this github repo, or using a specific wiki.hd-zero.com url and a framework like [mkdocs-material](https://squidfunk.github.io/mkdocs-material/getting-started/).

MartinHugh, 2023-06-02:
There is so much more to do in this area.  By making it clearer to end users, Divimath could reduce support tickets.  There is currently no clear way of self-diagnosing problems.  I currently have a brand new vtxs on the latest firmware which shows a blue flashing LED after start up; despite there being adequate cooling, despite connecting to different cameras and MIPIs. 

Surely it would be in everyones' interest to make diagnosis easier.  The current situation for doing this is poor and disappointing. How hard can it be?

pitts-mo, 2023-07-16:
Adding information about the PAT test and some of the current LED behaviors to the README.md would be helpful.

AlexandreBonneau, 2023-09-21:
This is now closed, but what has been changed if you mind me asking?

ligenxxxx, 2023-09-21:
@AlexandreBonneau
Regarding the diagnosis without video, it may be:
- Bad hardware.
- Bad settings.
- Bad firmware.
I agree with modifying the way the LED blinks, but I want to make a complete instruction and troubleshoot first. When everyone knows how to query information about hdzero, then modify the LED status. Otherwise, suddenly changing the current LED logic may make many people feel overwhelmed. We first need a manual that changes with the development of fw.
For PAT test, I don’t want to introduce it in release fw, mainly for stability.

MartinHugh, 2023-12-30:
One of the fundamental problems with diagnosing the existing diagnostic codes (not including LED off, for no camera) is that it requires that the user can tell the difference between 1/2/4 hz.  A lot of people cannot do this without recording the pattern and then measuring it in software video tools.   [I note that the recent addition of "dm6300 lost" breaks this trend of displaying a 50:50 mark:space at a given frequency, which is a good improvement, but this does not help the other diagnostics.

A much more intuitive way to do this is to go the route of ESC beeping which use long and short pulses.  One limitation in the vtx is that it is based around a 1s timeframe, divided into 1/16s intervals.   To get a really extended set of diagnostics this frame should be extended, but even working with a 1s time frame it is easy to see how we can get 6 or 7 diagnostics codes which are much easier to distinguish between.

[image]

This would help today, and can be extended upon later by increasing the time frame.

So rather than saying :

Diagnostic code A = 1Khz
Diagnostic code B = 2Khz
etc

we would be able to document
Diagnostic code A = Long
Diagnostic code B = Long - Short
Diagnostic code C = Long - Short - Short
Diagnostic code D = Long - Long
Diagnostic code E = Long - Long - Short
Diagnostic code F = Long - Long - Short - Short
Diagnostic code G = Long - Long - Long

Diagnostic code H = and so on

MartinHugh, 2024-02-05:
https://github.com/hd-zero/hdzero-vtx/pull/187

MartinHugh, 2024-06-06:
This issue was satisfied in [PR 187](https://github.com/hd-zero/hdzero-vtx/pull/187)

closed 2024-06-06 by MartinHugh