hd-zero/hdzero-vtx #128: [Feature Request] Better pitmode

lesharodin, 2023-06-02:
Since https://github.com/hd-zero/hdzero-vtx/pull/117 doesnt work anykind and I dont know why its merged..
Can we instead have an option to chose pitmode default power?
For example:
Team Race = OFF - Default pitmode (idk its real power but its powerfull enough to fly 100m)
Team Race = ON - Default pitmode is 0mw, and then we  dont need paralize state from FC, just make PIT+Paralize in modes and your stuck on race quad can be paralized as it should be.

Also PITMODE now have some bugs where on 1/5 runs VTX is stuck in pit even after arm. (osd says its not in pitmode, but it is, and only repovering helps). Not suitable for race events where you have your quad in pitmode all the time before arming.

AlexandreBonneau, 2023-06-02:
Somehow in my hdzero-converted fleet I have 2 drones that output no hdzero signal, while the red and blue leds are solid.
Using the vtx on/off stick movements does not activate the signal. On my drone with a buzzer I do heard a quick small beep when using ↙️ ↘️ , but that's it.

With my race v2 the problem started when I used the stick movement after having crashed to disable temporarily the VTX, but then I could not get the signal back. At one point I'm not sure how I managed that but with that race v2 VTX I could get a very faint signal on my bench, and I quickly disabled the P1MW mode, but it seems the setting has not been saved since I've had no luck getting the signal back since then.

On my second quad (fresh new build), the MSP port is configured with the correct UART, but again no signal is found by the HDZero goggles.
Just to be sure, when I plug my Tinytrainer quad, the HDZero is picked up right away so it's not the goggles.

According to discord, many users have had problems with the VTX either not outputting any signal, or outputting the wrong power level, or (like me) having the power level not saved correctly between power cycles.
One user claims having found the cause for the latter problem, and a workaround:
- Problem: It seems every battery cycle shifts to the next powerlevel
- Workaround: He sets 2 power level (25mW and 1W) on a switch, then before a flight use this switch twice to get 1W

If the powerlevel cycle problem is real, then perhaps it could explain why I get no signal while being stick in pitmode?

ligenxxxx, 2023-09-19:
@AlexandreBonneau Can you still reproduce this bug now?