hd-zero/hdzero-vtx #133: [Bug] LP Mode, Pit Mode, Vtx OFF on first boot not working on INAV 6.0 and FTTEC Alpha

T-M-0, 2023-06-09:
The three Modes, LP Mode, Pit Mode, Vtx OFF on first boot do not working on INAV 6.0 and FETTEC Alpha.
On both the MSP communication is enabled for OSD and Stick commands (FC and Vtx Menu)

Tested it with older firmwares from FETTEC Alpha and the actual INAV 6.0. Its always the same behavior. 

This issue is since Vtx Firmware Rev 06012023

ligenxxxx, 2023-09-21:
@T-M-0 https://github.com/hd-zero/hdzero-vtx/actions/runs/6068503029
Please test this fw.

T-M-0, 2023-09-23:
It did not fix the issue.

I tested it with:
Freestyle VTX on Inav 6.11 and FETTEC Alpha
Race V2 VTX on Fettec Alpha

@ligenxxxx do you have a FETTEC FC? If not, maybe I can arrange something

ligenxxxx, 2023-09-25:
I will test it with iNav6.x.x.

Thanks.

ligenxxxx, 2023-09-26:
@T-M-0 https://github.com/hd-zero/hdzero-vtx/actions/runs/6307941869
I reproduced and fixed it using iNav FC.
Please check this.

T-M-0, 2023-09-26:
The Stickkommand to switch the VTX off work.
You are the man 💯 👍 
I didn´t test the other modes.
One mentioned one thing is, with this firmware. Some OSD items are missing.

I tested it with:
Freestyle VTX on Inav 6.11 and FETTEC Alpha
Race V2 VTX on Fettec Alpha

closed 2023-10-12 by ligenxxxx