hd-zero/hdzero-vtx #136: HD OSD elements off-screen

evilC, 2023-06-17:
I originally posted this on the [BetaFlight Github issue tracker](https://github.com/betaflight/betaflight/issues/12822), but it's seeming likely that this is an HDZero issue...  

### Describe the bug

When I flash BF 4.4.1, it appears as if the coordinate system of the OSD is wrong - elements are off-screen to the right and bottom.  
This affects both the HUD elements, and menus.  
On one quad (The GHF405), flashing back to 4.4.0 solved the issue. I then flashed back to 4.4.1 and the issue came back. I then set the port to nothing, saved, then back to MSP+Displayport, and the issue went away.
On the GHF411 quad, this "turn it off and on again" trick did not work.

In the below image, I have moved the OSD elements to the top left quadrant of the screen to compensate for this issue, but as you can see, on the actual OSD they are filling the screen.
This is a screen grab from my DVR - I am plugged into USB with the configurator open (So you can see my OSD settings), and recording the screen with my quad
[image]

### To Reproduce

FW: BF 4.4.1  
HD OSD (I am using HDZero)

### Expected behavior

OSD elements should be on-screen

### Support ID

```txt
Id: "9f9ddcd8-110d-4a83-8c3d-0fab939feb3f"
```

### Flight controller

JHEMCU GHF411Pro AIO MPU / JHEMCU GHF405Pro AIO MPU

### Other components

HDZero whoop lite VTX  (FW v1.2.0)
HDZero nano lite camera
720p60 mode
HDZero Goggles (app 9.0.12 rx:0 va:161)

### How are the different components wired up (including port information)

GHF411: VTX on UART 1
GHF405: VTX on UART 2

lesharodin, 2023-06-23:
Try to set osd to pal then to hd again.
It helps with first setup when this bug appears.
It sometime can appear after that randomly, but then just reboot of FC resolve itself

ligenxxxx, 2023-06-25:
> Try to set osd to pal then to hd again. It helps with first setup when this bug appears. It sometime can appear after that randomly, but then just reboot of FC resolve itself

For betaflight, hdzero will force enable HD OSD on power-up, so setting pal and ntsc may have no effect.

lesharodin, 2023-06-25:
> > Try to set osd to pal then to hd again. It helps with first setup when this bug appears. It sometime can appear after that randomly, but then just reboot of FC resolve itself
> 
> For betaflight, hdzero will force enable HD OSD on power-up, so setting pal and ntsc may have no effect.

It still helps. Some rangom bug maybe

evilC, 2023-07-01:
Seeing more and more of this popping up on facebook and other places.
The common theme seems to be HDZero goggles (Not seen any reports of it happening with a VRX yet)  
It never used to happen, so this leads me to believe it is due to a change in something somewhere, however it is entirely possible that it could be due to a change in BF I guess.

tkrrrr, 2023-07-09:
I experienced the same issue. It appeared after I flashed a new build to enable another feature. I don´t remember the previous version and build options though. After some attempts I got it to work again when I flashed with VTX and SD-OSD enabled in the cloud build options. I previously had at least the SD-OSD disabled. Maybe it was just a coincidence / I didn´t flash again to check if I can reproduce the behaviour / isolate the relevant option, but maybe it helps.

Cightline, 2023-08-04:
> Not seen any reports of it happening with a VRX yet) 

It happened on my 4SMA VRX a little while back after updating the VTX/VRX, I downgraded the VRX only and everything is back to normal. I'm using Arducopter.

kaack, 2023-08-07:
I ran into the same issue today immediately flashing Betaflight 4.4.1 as  well. I was able to workaround it by moving some OSD elements around the screen, and setting to PAL, then back to HD (required multiple reboots of the goggles, and the vtx)

matyas-brenner, 2023-09-12:
I can confirm this issue with HdZero whoop lite vtx and BF 4.4.2.
https://www.facebook.com/groups/hdzero/permalink/681293100565321/

matyas-brenner, 2023-09-13:
The easiest workaround (maybe even fix, with updated Presets), IMHO is to cut out the MSP initialization, which feeds the FC VTX table and OSD canvas size and forces it to be HD. That can be all set up manually, which leads to more complicated configuration and more versatility and freedom to end users.

matyas-brenner, 2023-09-16:
I managed to workaround this by hardcoding desired resolution and FC version, which proves my original hypothesis that there is something wrong with initialization of the MSP. Check this commit for more details: https://github.com/matyas-brenner/hdzero-vtx/commit/aee8d4148b1ce8269a52ef4094c7563003182892

It is still unknown to me, if it really is a HdZero issue, or BF... Never the less. The initialization doesn't work reliably as it is.

I have some ideas about where the bug may be on HdZero side:
1. It seems to me, that the MSP configuration is quite fragile. There is no handling of messages that won't pass CRC check. In that case if the initialization messages don't go through, bad things can happen.
2. There can be some race conditions about the handling of the messages. As written, FC identification must come from FC first and then the MSP stuff. IMHO, there is no guarantee of that.

I'm, sadly, not equipped to investigate this further and the workaround I made fixes the issue for me. It is certainly not "universal bug fix", since it will work only with HD resolution 50x18 and Betaflight. Maybe it will give someone else the idea where to start with a proper fix.

Just a note: I'm sorry to say, but this project may be open source in terms of its openness of the code, but it is certainly not very forthcoming to devs. There is zero to little documentation and the code is IMHO not well-written. It uses a lot of passing values through global variables, there is nothing commented, a lot of "magical consts". It is not easy to understand how it works. There are also none commit descriptions. I, as an author of the code, would be lost in it in short time. I'd love to see this improved.

Don't take me wrong... it is great that HdZero decided to be an open source. Thank you very much for this, original authors.

evilC, 2023-09-17:
I am not wholly sure what you are referring to in that commit, but from a quick glance at the MSP protocol (And please forgive my ignorance if I got it completely wrong)...
The `command` type (Which appears to be where something initiates a communication and passes a parameter) - ie FC saying to VTX "You need to be set up thus" has no response - it appears to be one way, similar to SA.
So it strikes me that it's entirely likely that the problem is that the FC is trying to pass configuration to the VTX, but the VTX never responds even if it works, so the FC has no easy way of knowing? Maybe the problem is with BF MSP-DisplayPort? ie BF should be sending VTX config data, and then requesting VTX config data, checking that they match, and if not, then try again?

matyas-brenner, 2023-09-17:
> I am not wholly sure what you are referring to in that commit, but from a quick glance at the MSP protocol (And please forgive my ignorance if I got it completely wrong)... The `command` type (Which [...]

I also may be incorrect, since I have no proper way of debugging the issue. Nevertheless this is how I understand it. Let's make this as constructive as possible. :-)

The next is about the inialization, not about the OSD drawing.

1. VTX send some requests to FC (`MSP_CMD_FC_VARIANT_BYTE`, `MSP_CMD_STATUS_BYTE`, `MSP_CMD_RC_BYTE`, `MSP_CMD_VTX_CONFIG_BYTE`) meaning "who are you? (Betaflight, iNav, ...)", "how are you doing?" "how is power, band and channel set?", not sure about the last one, actually
2. The VTX is listening for the replies 
3. When `MSP_CMD_FC_VARIANT_BYTE` reply comes, the VTX will make save it. When this is lost (either not sent or CRC mismatch), we are done and the inialization will fail, since it is FC type dependant. That is the firs thing, I changed. I hardcoded it to be BTFL.
4. Then we are waiting for `MSP_CMD_DISPLAYPORT_BYTE` and it's subcommands. I have no idea why this comes. If it is a result of one of the requests, or if the FC is supposed to send it because MSP is configured on a corresponding UART. One of the subcommands in the body of this message is `SUBCMD_CONFIG`, which tells the VTX how is the OSD mode set up (HD/SD). The default of the VTX is SD, so when this is lost (either not sent or CRC mismatch), the FC "talks HD" and the VTX "listens SD", which IMHO leads to the OSD off screen, because the FC assumes that the canvas is bigger than what VTX things it to be. There is different initialization for iNAV, which means that even when this would be processed correctly, it will not be, if the VTX doen't know that BTFL is talking. So if 1. fails, this will never happen and thus fails too.
What I have done here is that I have again hardcoded that the resolution is HD_5018 and that this command is ignored.
5. Even if the replies would be OK and come in the other order, the initialization will fail, because in point 4. we need to know that BTFL is talking. - I'm not 100% sure about this.
6. The FC may lose the initial commands send in 1. because of a CRC mismatch.
Just a side note. As part of the initialization, the VTX is sending to FC the VTX table, it's canvas size. This is not the problem in my case. according to CLI in BF (diff, dump) it is set up correctly.

So to sum it up. The problem is definitely communication between FC and VTX via MSP. Since MSP implementation in BF is older and as such more tested, it seems that the probability of a bug on HdZero VTX side is more likely but not certain. 

What I have done is not a fix, just a hardcoding the configuration that VTX should have read in _my_ case (certainly not everybody's - especially if using SD OSD or not using BTFL), but for what ever reason (described here or not) didn't.

evilC, 2023-09-17:
Sounds like it's maybe something along the lines of what I said, but maybe with the roles reversed?  
ie the VTX on startup asks BF whether to use HD or SD, and in the case that the FC does not reply, or the CRC is bad, or some previous request that was meant to go OK had failed, it just assumes SD and does not re-request?

matyas-brenner, 2023-09-17:
> Sounds like it's maybe something along the lines of what I said, but maybe with the roles reversed? ie the VTX on startup asks BF whether to use HD or SD, and in the case that the FC does not reply, [...]

Yes, that is what I thing of it.

What I can't say for sure is, who's to blame - FC or VTX. I don't fully understand how the commands work and which side is supposed to send what and which side is responsible for re-send if needed. I assume that vtx side is to blame here, since it is the less mature of the two and I was also unable to find this undesired behavior with walksnail. Since it uses the same MSP protocol in the same situation. To be fair, I'm not using WS and I'm not a member of it's FB, discord or any other group, so I may have missed a lot.

ligenxxxx, 2023-09-19:
@matyas-brenner How can I reproduce it?

ligenxxxx, 2023-09-19:
I did some tests and the environment is as follows:
- betaflight fc 4.4.2
- hdzero vtx 1.4.0

How to reproduce the error:
- Set betaflight osd format to pal or ntsc
- Restart quad
- The configurator shows that the osd format is HD, but the goggle actually shows it as pal/ntsc.
- Restart quad
- goggle display HD osd.

I feel like it's a problem with hdzero vtx.

matyas-brenner, 2023-09-19:
> @matyas-brenner How can I reproduce it?

@ligenxxxx, thank you for looking into this.

This happens to me on multiple quads one of them is "standard" Mobula 6 with HdZero. The only modification, I've done is to replace pigtail for ph2.0. Diff will follow, but I thing that the only important things are MSP+VTX and HD OSD set. I'm using Betaflight 4.4.2 and 4.4.1 also tested (the same results - these versions have the same version of MSP API, so no surprise), VTX version 1.4.0 and VRX4, but I can assume that all VTX versions supporting HD OSD are affected and also all of the VTXes, since this code is shared.

The FC is set, from the beginning to use HD OSD. Just power cycle the quad several times (many in some cases) sometimes the OSD is right and sometimes it is not.

If I would to debug this, I'd use a "virtual FC", just a simulated UART to which I'd write data myself, and try to send the relevant commands with intentionally corrupted CRC or to replace the order of the replies. I would also check based on which the FC sends `MSP_CMD_DISPLAYPORT_BYTE` and if it is send in all cases, but you probably know that as the original author.

I'd maybe start just by writing the received messages instead of OSD on the screen and just attempt to fly, since there is probably no storage to log the messages into.

Here is the diff:
Just a side note. Lowband is wrongly named because this dump was taken before the vtx was updated to 1.4.0.
```
# diff all

# version
# Betaflight / STM32F411 (S411) 4.4.2 Jun  1 2023 / 21:36:21 (e43d591b2) MSP API: 1.45

# config: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CRAZYBEEF4SX1280
manufacturer_id HAMO
mcu_id 005800375047500520333539
signature 

# name: MB

# feature
feature -RX_SERIAL
feature -SOFTSERIAL
feature -AIRMODE
feature RX_SPI

# serial
serial 1 131073 115200 57600 0 115200

# beeper
beeper -ARMED

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1750 2100 0 0
aux 1 1 2 1700 2100 0 0
aux 2 2 2 1300 1700 0 0
aux 3 13 4 1700 2100 0 0
aux 4 28 1 1700 2100 0 0
aux 5 35 4 1300 1700 0 0
aux 6 36 3 1700 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY    0    0    0    0    0    0    0    0
vtxtable band 2 BOSCAM_B B FACTORY    0    0    0    0    0    0    0    0
vtxtable band 3 BOSCAM_E E FACTORY    0    0    0    0    0    0    0    0
vtxtable band 4 FATSHARK F FACTORY    0 5760    0 5800    0    0    0    0
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I FACTORY    0    0    0    0    0    0    0    0
vtxtable powerlevels 3
vtxtable powervalues 14 23 0
vtxtable powerlabels 25 200 0

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_type = PT2
set gyro_lpf2_static_hz = 375
set dyn_notch_count = 1
set dyn_notch_q = 500
set gyro_lpf1_dyn_min_hz = 187
set gyro_lpf1_dyn_max_hz = 375
set acc_calibration = -11,61,-17,1
set min_check = 1010
set max_check = 1990
set rc_smoothing_auto_factor = 52
set blackbox_device = NONE
set dshot_idle_value = 600
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 300
set vbat_warning_cell_voltage = 330
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set simplified_gyro_filter_multiplier = 75
set report_cell_voltage = ON
set osd_warn_bitmask = 8061
set osd_rssi_alarm = 40
set osd_vbat_pos = 407
set osd_rssi_pos = 34
set osd_link_quality_pos = 14337
set osd_link_tx_power_pos = 2081
set osd_rssi_dbm_pos = 377
set osd_tim_1_pos = 54
set osd_tim_2_pos = 14816
set osd_flymode_pos = 15374
set osd_g_force_pos = 65
set osd_throttle_pos = 14848
set osd_vtx_channel_pos = 399
set osd_crosshairs_pos = 237
set osd_ah_pos = 110
set osd_current_pos = 14880
set osd_mah_drawn_pos = 15884
set osd_craft_name_pos = 8426
set osd_gps_speed_pos = 409
set osd_gps_lon_pos = 97
set osd_gps_lat_pos = 129
set osd_gps_sats_pos = 32
set osd_home_dir_pos = 12
set osd_home_dist_pos = 14
set osd_altitude_pos = 407
set osd_warnings_pos = 14769
set osd_avg_cell_voltage_pos = 15916
set osd_disarmed_pos = 14741
set osd_flip_arrow_pos = 14488
set osd_stat_bitmask = 52574060
set osd_displayport_device = MSP
set osd_canvas_width = 50
set osd_canvas_height = 18
set vtx_band = 5
set vtx_channel = 7
set vtx_power = 2
set vtx_freq = 5880
set vcd_video_system = HD
set gyro_1_sensor_align = CW0
set gyro_1_align_yaw = 0
set expresslrs_uid = <my_uid>
set expresslrs_rate_index = 1
set craft_name = MB
set pilot_name = MB

profile 0

profile 1

profile 2

# profile 2
set dterm_lpf1_dyn_min_hz = 101
set dterm_lpf1_dyn_max_hz = 202
set dterm_lpf1_static_hz = 101
set dterm_lpf2_static_hz = 202
set crash_recovery = ON
set iterm_relax_cutoff = 20
set p_pitch = 66
set i_pitch = 77
set d_pitch = 61
set f_pitch = 160
set p_roll = 67
set i_roll = 78
set d_roll = 60
set f_roll = 162
set p_yaw = 67
set i_yaw = 78
set f_yaw = 162
set d_min_roll = 60
set d_min_pitch = 61
set d_max_advance = 0
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 45
set feedforward_jitter_factor = 8
set feedforward_max_rate_limit = 95
set simplified_i_gain = 65
set simplified_d_gain = 200
set simplified_pi_gain = 150
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 135
set simplified_pitch_d_gain = 90
set simplified_pitch_pi_gain = 95
set simplified_dterm_filter_multiplier = 135
set tpa_breakpoint = 1250

profile 3

# restore original profile selection
profile 2

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 127
set pitch_rc_rate = 127
set yaw_rc_rate = 127
set roll_expo = 40
set pitch_expo = 40
set yaw_expo = 40
set roll_srate = 72
set pitch_srate = 72
set yaw_srate = 72

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

lesharodin, 2023-09-19:
@matyas-brenner did you try to build fw with only HD osd (without SD osd)?
Seems like it helps me.

matyas-brenner, 2023-09-19:
> @matyas-brenner did you try to build fw with only HD osd (without SD osd)? Seems like it helps me.

No, I didn't. I needed a quick fix before a flying event with as few unknown stuff that could go wrong as possible. I'll try it and report back here, but it will take some time, certainly not today, sadly.

I assume that you mean only with HD OSD hardcoded, which leaves FC to be determined "normally" via MSP communication.

lesharodin, 2023-09-19:
@matyas-brenner 
[image]
Yes, just disable SD OSD in build options

matyas-brenner, 2023-09-19:
> @matyas-brenner [image] Yes, just disable SD OSD in build options

What is the point? It is already proven, that the FC is correctly using HD OSD as it should, removing the SD capability from the build makes no difference than.
The VTX is the one assuming SD OSD. It is already proven that the mistake is in the MSP communication.

lesharodin, 2023-09-19:
@matyas-brenner I just stop facing this issue when I start building BF with only HD osd.
I dont say its 100% fix, but my friends who use standart BF build options facing this issue quiete often.

matyas-brenner, 2023-09-19:
> @matyas-brenner I just stop facing this issue when I start building BF with only HD osd. I dont say its 100% fix, but my friends who use standart BF build options facing this issue quiete often.

@lesharodin, interesting... I'll try it, but frankly I suspect it to be just a coincidence. - I'll try reflashing the VTX with original 1.4.0 and reflashing BF as suggested anyway.

ligenxxxx, 2023-09-19:
@matyas-brenner 
I recommend using a usb to serial port tool to capture the uart tx and rx data at the same time and analyze it when an error occurs. I can't reproduce the situation you mentioned (just restarted multiple times).

matyas-brenner, 2023-09-19:
> @matyas-brenner I recommend using two usb to serial port tools to capture the uart tx and rx data at the same time and analyze it when an error occurs. I can't reproduce the situation you mentioned (just restarted multiple times).

I'm sorry @ligenxxxx, I know that it is selfish of me, but I am not equipped to do so. I know that it is not expensive equipment. I made it work and I believe that for most of the users with the same issue, since I'm using Betaflight with HD OSD as most other HdZero user probably are. I also believe that my changes should not be in the official build since it limits some users.

Maybe in time, when my schedule will be more free, I'll find the time to debug it further but now I simply don't have the time (or more precisely, I prioritize other things).

I can't event say if it is a "hard bug" (BF or VTX), miscommunication due to electric noise, race condition of the communication, or whatever else. I see that as is, it "mostly works".

Now I have 4 HDZero quads (2x whoop lite, Race v2, freestyle v1 batch 1), one of the quad is bought as BNF - Mobula 6.
What I can safely say is, that it happens to all of them sometimes since the HD OSD was introduced. Each has different FC, each is built differently, each has different wire routing from trivial 1cm to really long underneath the stack. This doesn't seem to have any affect, but others reported, that changing wire routes helped.
I found that if it happens once, it is likely to happen on the next power up again, but even that I can't safely confirm. It may just be how probability works - sometimes you roll multiple 6's in a row and sometimes you don't. This issue happened to me on most of my flying days, but not every time.

I know that it is not much to start with... that is why I decided to hardcode the settings and not to deeply investigate it.

As it seems I'm not the only one with the same issue. It is reported by others, but to be fair the numbers are not great.

I'd love to see this fixed, as any other issue that effects the experience using HdZero, but if the fix for me is to merge "official" changes to my fork, where I have commented like 5 lines of code and made 2 things constant... I'm OK with that.

ligenxxxx, 2023-09-19:
@matyas-brenner 
HDZero needs to support both HD OSD and SD OSD. Because users not only use the latest betaflight firmware, but may also use other flight controllers or older versions, we need to be compatible with old functions.
That's okay, I have a lot of time to test the firmware and look at everyone's issues on the web. When I catch a bug, I destroy it.

ligenxxxx, 2023-09-21:
@evilC @matyas-brenner I made some modifications that should help with this.

In the past:

When betaflight defaults to HD osd:
- VTX will send MSP_SET_OSD_CANVAS(188) to FC once.
- VTX waits for FC to respond, but FC never responds because betaflight sees no change in OSD resolution and then responds to VTX.
- VTX changes its resolution to 5018 after receiving the response.

When betaflight is not HD osd by default:
- VTX will send MSP_SET_OSD_CANVAS(188) to FC once.
- VTX waits for FC to respond, but FC never responds because betaflight found that the OSD resolution changed and then saved the parameters and then restarted immediately. So VTX osd always works in SD mode.

Now:
- VTX will send MSP_SET_OSD_CANVAS(188) to FC once.
- VTX sends MSP_GET_OSD_CANVAS(189) in 8HZ loop.
- VTX checks if HD OSD is enabled after receiving MSP_GET_OSD_CANVAS response. And dynamically responds to FC OSD resolution, so it has nothing to do with whether FC is reset.

closed 2023-09-25 by ligenxxxx

evilC, 2023-11-01:
Many thanks @ligenxxxx, I will flash this and test this weekend :)

evilC, 2023-11-01:
As a side thought, is there anything in place to warn the user that this step is failing: "VTX sends MSP_GET_OSD_CANVAS(189) in 8HZ loop". One would imagine that there is some kind of timeout, or does it keep trying forever?
In the instance that the FC RX <--> VTX TX connection is not working (ie broken wire / [broken FC RX](https://github.com/hd-zero/hdzero-vtx/issues/127) pad etc), it would be nice to have some kind of warning, particularly for people who are not familiar with these issues.
A similar thing may also apply WRT LP mode. Does this now require two-way comms to properly work? If so, a similar warning may also help?

evilC, 2024-12-30:
It's maybe worth mentioning, for anyone stumbling upon this in the future - if you think you have this issue - check that the RX pad of the FC is correctly connected to the VTX. If it isn't, you can also see symptoms similar to this. Your VTX menu (Sticks down and together) will probably not work too