hd-zero/hdzero-vtx #152: PAT Version of firmware

zandorsp, 2023-08-27:
In the Readme.md ttroubleshooting section there are a referece for the PAT (pattern?) version in :

> 
>
> use a PAT version of firmware for your specific VTX to verify RF output from VTX and test patern signal is received by VRX.

But no indication to where to download it.

pitts-mo, 2023-08-27:
You should be able to compile with the PAT define in platfrmio.

If you need one quickly you can request a PAT version for your model VTX
from
***@***.*** or maybe somone from discord server can build it for
you.

-p

zandorsp, 2023-08-27:
> #define VIDEO_PAT
> // #define FIX_EEP

> You should be able to compile with the PAT define in platfrmio. If you need one quickly you can request a PAT version for your model VTX from ***@***.*** or maybe somone from discord server can build it for you.
> […](#)
> -p

is in common.h uncommenting  #define VIDEO_PAT . I'm right?
>// mode
>// #define _RF_CALIB
/>/ #define REV_UART

closed 2023-10-31 by zandorsp