hd-zero/hdzero-vtx #165: MSP VTX OSD channel designator changes at arming - very confusing

MrMap, 2023-10-16:
This was originally filed by me on INAV Github issue tracker. Somebody told me this is a HDZero bug - hence a repost here on HDZ Github.

Original INAV post: https://github.com/iNavFlight/inav/issues/9374
Please note: The YouTube link below provides a clear demonstration of the issue.

Steps to Reproduce
Start out with e.g. Raceband Ch5 Pwr 2, using the EDGETX ELRS VTX LUA, confirm OSD display correct CH: R5 : 2
Arm (Aux 1 / CH5)
Watch the OSD changes to CH: A1: 2
**_The VTX Channel in use has not changed_**, it will still be on R5

Additional context
I use HDZero goggles, all FW 22082023
HDZ Freestyle v1, current FW _22082023

Here is a short DVR video demonstrating the phenomena.
https://youtu.be/1cz541mFGg8

FC Board name and vendor:
version
INAV/SPEEDYBEEF405WING 7.0.0 Sep 11 2023 / 12:56:24 (https://github.com/iNavFlight/inav/commit/3a2412e5e680da0460f54f482d23e88e8ef128c1)
GCC-10.3.1 20210824 (release)

ligenxxxx, 2023-10-25:
@MrMap Did you use Smartaudio?

MrMap, 2023-10-25:
No, I disabled SA since I transitioned to INAV 7 and MSP VTX (this was my only reason for INAV 7)
Only use MSP VTX, much more reliable than SA, though maybe still some minor snags (difficult for me to say if in INAV 7 or HDZ VTX FW - or in ELRS backpack / LUA).

ligenxxxx, 2023-11-02:
@MrMap Please test 1.5.0-rc1

MrMap, 2023-11-02:
@ligenxxxx Will do!

closed 2023-11-06 by ligenxxxx