hd-zero/hdzero-vtx #169: Lowband on v1.5.0-rc1

zandorsp, 2023-10-30:
To reproduce:
Update a lowband unlocked vtx with v1.5.0-rc1 (tested on Whoop vtx and Race v2).

The lowband channels (L1-L8) doesn't work anymore.

Downgraded on v1.4 and the lowband channels worked again.

ligenxxxx, 2023-11-01:
@zandorsp I use Race v3 VTX and enable msp vtx. After updating to v1.5.0-rc1, lowband is still valid in the VTX menu.
Do you use power on switch or cms(fc) menu?

ligenxxxx, 2023-11-01:
@zandorsp Please test this fw.
https://github.com/ligenxxxx/hdzero-vtx/actions/runs/6717513533

zandorsp, 2023-11-01:
> @zandorsp I use Race v3 VTX and enable msp vtx. After updating to v1.5.0-rc1, lowband is still valid in the VTX menu. Do you use power on switch or cms(fc) menu?

To chance channel I used the ExpressLRS VRX menu:
[image]

But I tried on menu too...

zandorsp, 2023-11-01:
> @zandorsp Please test this fw. https://github.com/ligenxxxx/hdzero-vtx/actions/runs/6717513533

Ok I flashed it and seen all right now.

closed 2023-11-06 by ligenxxxx