hd-zero/hdzero-vtx #171: Reset LowBand or 90FPS mode using firmware update?

evilC, 2023-11-01:
It seems to me that there is a capacity for people to get locked out of a VTX because LowBand or 90FPS is set and they only have a VRX.
Am I missing something, or is there a way out of this scenario?
If there is not, would it be possible to craft a firmware which resets the VTX to some safe defaults?

ligenxxxx, 2023-11-06:
https://github.com/ligenxxxx/hdzero-vtx/actions/runs/6243947339

zandorsp, 2023-11-16:
Now with vrx's firmware rev 13112023 with narrowband support this is still necessary?

evilC, 2023-11-16:
I guess maybe not, but I did see recently so that there is actually a firmware which does do it anyway

ligenxxxx, 2023-11-29:
Updating the firmware will not affect the parameters that have been saved. Unless special firmware is used, saved parameters are initialized at power-up.
[image]
Just `#define RESET_CONFIG` to generate the special firmware.

MartinHugh, 2024-01-03:
I have seen a number of users locked into narrow band without realising it, and assuming they have a defective vtx.

Would it be possible to add the ability to switch in and out of narrow band by stick commands ?

This would avoid the need to recompile and reflash firmware as part of diagnostics.

ligenxxxx, 2024-01-03:
https://hd-zero.github.io/hdzero-vtx-docs/vtx_user_manual/#stick-command
How do you think we should design a new stick command to reset all configuration?

MartinHugh, 2024-01-03:
> @MartinHugh https://hd-zero.github.io/hdzero-vtx-docs/vtx_user_manual/#stick-command How do you think we should design a new stick command to reset all configuration?

I realise there are a number of challenges here, but I think the demand has increased judging by the number of people who seem to be unwittingly locked into narrow band.

One good reason *not* to do this with sticks is that we really do not want to allow people to accidentally reset all there settings.
Another issue is that most/all movements are already employed in iNav/Betaflight/HDZero positions.

The only real way would be to modulate them differently.
I note that Quicksilver uses a sequence of 3 movements in a certain direction for example.

I understand stick movement would be a challenge, but raise this point to try and get some discussion of some way to resolve 
the underlying problem.