hd-zero/hdzero-vtx #182: Signal spills over into other channels

SentinelFour, 2023-12-09:
Hi, I just set up a new RotorHazard and did the node configuration. I noticed that there was (significant) noise on other channels/frequencies. I thought it was a problem with my node or that the VTX was broken. I tried different HDZero VTXs (whoop and whoop lite, race v3) and they all behaved the same. 

Also, all of my other nodes when switched to R6 detected the same noise:

[image]

When I reached out to other people on the discord, they confirmed similar observations, with some race directors limiting the number of analog pilots if there are too many HDZero pilots participating.

Is this something HDZero is aware of or working on? When set to R1, R6 has the most noise, but other channels are affected as well. When set to a different channel, there is always at least one other channel that has noise... All tests with 25mw.

ligenxxxx, 2023-12-14:
I don't know what causes this and I don't know how RotorHazard works, but I can attest that the HDZero VTX outputs clean RF.
I did a small test using an RF spectrum analyzer. I used whoop vtx, 25mw. Switching from R1 to F4 respectively, I can't see any interference signals in other channels.
https://youtu.be/IteFesP6aAA?si=iYlVVgR8Oz4qvSNb

fpvdoc, 2023-12-14:
I just finished testing 7 different quads and ended up with two of them spraying into R8 when set to R3. One of them is an old v2. Thus I thought it might be wear but the second quad showing the same spraying is a brand new freestyle VTX v2 set to 25mW even on different quality antennas like RushFPV or TBS Triumph.

SentinelFour, 2023-12-14:
Thanks @ligenxxxx  for taking the time to look into this. Also for doing some real tests with more professional equipment, that's for sure. I can say that I can reproduce this issue on 3 out of 4 quads. The fourth (not originally mentioned above because it wasn't ready at the time) doesn't have the problem - it's a used Whoop board. Analog quads with AIO boards, TBS unifies, did not have this behavior.
All HDZero VTXs are running firmware 1.5.0. I restarted RotorHazard in between, but in my case the results did not change. I also tried different antennas. The whip antenna that comes with the whoop kit, other whip antennas, Foxeer and Axii stubby. No change.

Not sure if this is a bug with my setup, with HDZero, or with RotorHazard. Based on my individual tests, HDZero seemed to be the most likely common denominator. If you want to close this issue and wait for more people (pilots, race director) to mention something in the future, I can understand. Otherwise, if there is anything I can do to provide you with more data, please let me know. Happy to help where I can :-)

fpvdoc, 2023-12-14:
Yesterday testing left me with 2/7 quads showing the "bleeding". Today I did some more testing and ended up with 4/7. Long story short, this is related to Pit Mode being on or off, with the quad like one meter in front of the tracker. I thought Pitmode would be applied automatically at quad boot up, if set to ON via LUA but this isn't case.

fpvdoc, 2023-12-16:
After some more testing I think this needs to be divided into two. First is the tracker showing some noise from other channels but this is neglectable cause it‘s far away from creating a Peak on a different Node/Channel.

 Three different analogue VTX behaving pretty much the same as HDZero VTX‘s regarding that.

The second issue I had, having Peaks on different Nodes is related to the order of the Arduinos being plugged in the Delta5 boards and/or how the two boards are connected via SDA/SCL among each other. I still couldn‘t reproduce the behaviour but I‘m condifent I will be able soon.

From my point of view this issue is in no way related to HDZero

SentinelFour, 2023-12-16:
> After some more testing I think this needs to be divided into two. First is the tracker showing some noise from other channels but this is neglectable cause it‘s far away from creating a Peak on a [...]

Thanks for the update and your thorough investor. I will also test more in the upcoming days just to see where it goes.
For your info, I have a nuclear hazard timer so no arduinos in it. Still, I can see the same pattern or issues. Couldn't recreate it on analog yet but didn't test all quads yet. If this indeed shows the same pattern, than this would clearly rule out Hdzero.

SentinelFour, 2023-12-17:
Hi, I tried to reproduce it more systematically, and frankly I have to admit that even the drones with HDZero were not causing the problem on my end today. Neither them nor any analog quad. I restarted all machines on day 1, including the tracker, and the problem only occurred with HDzero, so I assumed that was the cause. Now I am not so sure, as the same setup and configuration did not trigger this behavior today...
@ligenxxxx I think it would make sense to close this issue for now - until more data from others points it in a certain direction. Thanks for your support and sorry!

fpvdoc, 2023-12-17:
I reworked my hardware today and did a fresh install including PI OS. So far everything is working as expected on analogue as well as on HDZero. This issue can be closed from my point of view.

HazardCreative, 2024-01-22:
> I don't know what causes this and I don't know how RotorHazard works

To add some context, for the purposes of this signal display, RotorHazard is simply collecting the RSSI value from the RX5808 module and writing it to the graph. There is a simple median filter applied to suppress high-frequency variability.

A [small amount of pickup on other channels](https://github.com/hd-zero/hdzero-vtx/issues/182#issuecomment-1848429114) is normal even with analog, but something as large as seen on R6 is unusual.

If further testing does identify an issue on the timer side, please file an issue with RotorHazard directly. https://github.com/RotorHazard/RotorHazard/issues