hd-zero/hdzero-vtx #186: [Feature Request] MIPI CSI - ribbon connector for single board computers?

Arduhat, 2024-01-17:
There could be a whole new world of possibilities opened up for the HDZero protocol if we can connect the camera direct to a single board computer. If the video could be split from the vtx and transmitted to the pilot, as well as streamed via cable to a companion computer, an enormous amount of potential use cases could be unlocked. Is this possible?

plurf, 2024-01-18:
If you need new hardware/application from the hdzero tech, maybe you better contact directly divimath

ThatOneGuyGreggers, 2025-01-25:
Has there been any update on this front? I would love to be able to connect a vtx to to the Raspberry Pi DSI output to be able to do AR type waypoints for flying, maybe a friend/foe for paintball/airsoft, and video monitor for photogrammetry. Also would be able to use the Pi as a DVR for recording flights. So many possibilities!

NulliFieDFPV, 2025-01-26:
that would be a great feature! let it get real pls!

ThatOneGuyGreggers, 2025-01-27:
Three things I see are hang up.

- What is the specific secured mipi connector for the vtx?
- What is the pin out for the HDzero vtx mipi connector?
- According to the hdmi into VTX post apparently there is an encryption IC or something used between the camera and VTX.

bxff, 2025-01-27:
> Three things I see are hang up.
> 
> * What is the specific secured mipi connector for the vtx?
> * What is the pin out for the HDzero vtx mipi connector?
> * According to the hdmi into VTX post apparently there is an encryption IC or something used between the camera and VTX.

The MIPI connector pinout is available here (see first 2 pdfs): https://device.report/divimath

The MIPI initialisation code is available in this repo (see `Set_1080P30` and similar functions.)

The encryption IC exists on both the analog (composite) and digital cameras, HDZ doesn't seem to want to communicate with the OSS community on it's working and that's a huge problem. See: https://github.com/hd-zero/hdzero-vtx/issues/232

Similar work has been conducted on getting hdmi-in through the MIPI interface, this used the LP mode of MIPI interface which allowed for 720p60fps, whist a ISP (AP1302 on the camera side) spoof would allow for full 720p120fps, there a lot more information on their wtfos discord with some current work being done by other users too: https://github.com/fpv-wtf/margerine/issues/17 

AFAIA the spoofing was only possible with the help of getting firmware drivers off vista using wtfos, the same could be possible with hdzero.

ThatOneGuyGreggers, 2025-01-27:
> The encryption IC exists on both the analog (composite)

Damn... I was thinking about using this route last night as a proof of concept type setup.

Thank you for the info!