hd-zero/hdzero-vtx #203: Unable to display VTX Power on iNav OSD

MartinHugh, 2024-06-06:
Using 
iNav 7.1.1 
vtx firmware 1.6.0
Whoop vtx

The iNav "Video TX Power" OSD element always shows as 0 rather than 1 (25mW) or 2 (200mw)
This works fine in 1.5.0, but not in 1.6.0

I suspect this is to do with the initialisation between the FC and the VTX.

This functionality works OK up until commit cea7f2c. 
A side effect of the following commit (f267b74) was that no iNav OSD was displayed at all
When the OSD was next available (3816938), although the OSD was back, the VTX power now shows zero

[image]

sixtyfive, 2024-08-14:
This sounds similar to what I'm experiencing: OSD working fine, but no band, channel or output power shown (the latter being displayed a 0) and no output power change via radio switch possible, either.

geoffsim, 2024-11-27:
This has been fixed in iNav 8.

closed 2024-11-27 by MartinHugh