hd-zero/hdzero-vtx #205: Questions: FreestyleV2 Power Output with Keyboard Only

Brets0150, 2024-06-17:
So I have a FreestyleV2 in a RC car, without a flight controller. I applied the 1W unlock. I have the HDZero keyboard, but the keyboard only allows you to select a power output of "1" or "2". 

I have two questions.
1. What is the power output for the values "1" or "2" on a 1w unlocked VTX? I am thinking 1 == 25mw and 2 == max? Or is "2" hard coded to 200mw?
2. If the "2" value is hard coded to 200mw, can I hard coded to 1w?

I have been looking through the code and am finding some hints to my questions, but C is not my primary language. So I am not sure. 

Any help is appreciated.

ligenxxxx, 2024-08-06:
https://hd-zero.github.io/hdzero-vtx-docs/latest/vtx_user_manual/#unlock-rf-power-limit

Brets0150, 2024-08-07:
Hello, Thanks for the reply. My question is not about the 1W unlock process. My question is more about the keypad functionality when the 1W unlock is applied. I have reviewed the wiki and tried to read through the code, but I cannot see a straight answer.

With the 1W unlock applied, what power output values are applied if option "1" or option "2" are used?

ligenxxxx, 2024-08-09:
1: 25mW
2: 200mW
3: 500mW
4: 1W