hd-zero/hdzero-vtx #232: HDMI Input using hdzero vtx as DSLR camera monitor

shanggl, 2024-12-17:
hd-zero vtx using TC358746 toshiba's csi to paralle port bridge chip ,convert mipi csi yuv422 8-bit video stream to  paralle port for DM5860 chip , the isp  eg RUNCAM using  FH8323 is on the camera side , very similar to DJI vista air unit .
Is it possible to develop a bridge chip like[ hdmi conver board ](https://item.taobao.com/item.htm?abbucket=11&id=799446521268&ns=1&pisk=gJpKnYjq-V0HuOOz5vGGrZ24wqnDpBKE-e-bEab3NFL9VN2oY9cPeUQ9zwfhdw2Re3LyrUKrYg_WPUQktcDDYHWPFqb-nxxEajWoJeQ5ANi5ViVIj6apA9rCFq0mnxxeYTWW-Dy5M5N10NZCV_T5COslmTsQO8iO5gI4NyT5FcnO2geQPTw75NscmTaQOWs1fiIAVM15VliO4NsCFTG1ykQuAaeJZfqTdxwyj8wWBMC1sH_BvHilx6nevN25bdSKuZtCW8wR8g7o6hKnJ--GLI8AmeD_pEKBGUOXb4eARIx25G1gq8x9jeJAdK3LKGx9VIf9ImMVp9_BZ9-uD2x9pexeB3q79UdD8I5JF2eA5pRMLsJn5JBXgHXHNFMTpEsrKKvxcz4ckgVKXLNza6sarlymwZIekwSOocVga75Gb8VpEDP7NrycXZmiF7PPsgC..&priceTId=2150442117344247151636260e1a90&spm=a21n57.1.item.2.286c523cIqZOnn&utparam=%7B%22aplus_abtest%22%3A%22acc1b4d7ae6fcec43d8ab61b27de8d08%22%7D&xxc=taobaoSearch)
for hd-zero vtx for 1080p video input , then hd-zero vtx should be able used for DSLR Camera or BMPCC / SONY  Camera for video transmmit .
i'm working on it and get a colorbar video .

shanggl, 2024-12-19:
The "AN_DM5680_CAM.pdf" said that "Each HDZero compatible Camera has to integrate an encryption IC for authentication."   i send a mail for more detail info but no response at all

shanggl, 2024-12-20:
After very detaild enquire ，divimath reply " we can't provide any detail info of the encription ic ,sorry ."  

so , forget about it .  opensource ?  just joking

ThatOneGuyGreggers, 2025-01-25:
Instead of just an HDMI adapter how about a way to connect it to a SBC like a Raspberry Pi? Possibly through the DSI connector? Connecting to a Raspberry Pi it might be easier to spoof the encryption IC.