hd-zero/hdzero-vtx #233: Eco VTX No start with 1.7 Fw

lightface79, 2024-12-22:
Eco VTX 
Fw 1.7.0
Programer usb with hdzero programer Windows
Upgraded from 1.6.0

Problem.
Power on the quad.
VTX Red Solid Red, Blue Led Blinks 1 time only then stays off.
No signal out from vtx, does not show up in scan on cannels in goggles.
Do report not working in FC Betaflight, No comunication.

upgraded both my Eco to 1.7.0 via Usb programer.
Non working after this.
Downgraded   to 1.6.0 on both....    Everything works again =)

Hdzero Windows programer have the 1.7.0 as the latest FW for the Eco vtx.

nerdCopter, 2024-12-22:
stick with 1.6.0 or try a newer build from Github Actions: https://github.com/hd-zero/hdzero-vtx/actions?query=branch%3Amain+event%3Apush+

lightface79, 2024-12-22:
@nerdCopter 
Dont know how to do that with the github actions!
But if the eco vtx dont work with the 1.7.0fw it should be removed from the hdzero programer so nobody runs in to the same problems i did and waste 2 days of error serching. =)

Or a Warning that the 1.7.0 fw dont work with the eco.

nerdCopter, 2024-12-22:
click into the action links, look on the bottom for the .zip files to download. maybe a newer firmware fixes it.  yeah, bugged firmware don't seem to be removed from website.